4am-robotics/cob_control #32: make cob_collission_velocity_filter usable for non-rectangular footprints

ipa-fmw, 2015-04-23:
this is needed for MLR bases

ipa-fxm, 2015-09-04:
@ipa-mig @ipa-fmw 
any progress here?

ipa-fxm, 2017-01-25:
@ipa-mig 
This is getting urgent.
I can have a look at this, but I guess you (or a student) already spent some thoughts on this...or even started something...that I could use as a starting point.
Could we meet up for a quick exchange of ideas?

ipa-mig, 2017-01-25:
@ipa-fxm Thusday morning?

ipa-mig, 2017-01-27:
as discussed, for what you wwant to do, this is actually a set of changes. I addressed them in the respective issues:

1. adapt cob_footprint_observer to provide not a rectangular footprint, but a convex polygon: #131
1. develop node that checks  local costmap and provides "Safety Stop functionality" by disabling the respective drive commands in the twist_mux
1. adapt this node to properly handle the convex polygon footprint

about the last point:
In principle, the current node could be adapted to work with any convex polygon. however, I'm not quite sure how complex this will get.
In the end, the proper way to do this would be to rather create a node that uses a local planner to create reasonable velocity commands based on the command input and the costmaps...