4am-robotics/cob_control #52: Re-work cob_cartesian_controller

ipa-fxm, 2015-08-24:
ToDo's:
 - review and re-structure profile_generator library (unification, design pattern)
 - use Action interface of `cob_frame_tracker` rather than Service interface
 - rework/improve safety-/abortion criteria (remove from `cob_cartesian_controller`, improve in `cob_frame_tracker`)

@ipa-fxm-cm

ipa-fxm-cm, 2015-11-19:
-Restructured profile_generator with factory builder approach
-Matrix sort not longer necessary
-Implemented a method to achieve equal array sizes (Adjust array length function only used in a special case. (Check out the code comment) )
-Combined several functionalities and simplified the code structure
-Might fixed the "Square box is not really a Box" problem ( Can be checked after the ToDo from below is fixed)

ToDo:
-There's still a bug in the optimized trajectory interpolation ( Path too short for a given velocity and acceleration )

ipa-fxm, 2015-11-21:
I extracted everything related to additional/new Cartesian motion primitives to a new issue https://github.com/ipa320/cob_control/issues/69.

ipa-fxm, 2015-11-21:
Current progress is reviewed in https://github.com/ipa-fxm/cob_control/pull/9

ipa-fxm, 2015-12-15:
I extracted everything related to re-working the action interfaces and abortion criteria to a new issue #75.

ipa-fxm, 2015-12-15:
Fixed in #74 (after the latest changes from my `test_new_cartesian_controller` branch have been merged!)

All the other open bullets are minor, mostly done, not relevant anymore or to be marked as wontfix.

ipa-fxm-cm, 2015-12-15:
How can we guarantee that the end-effector is at the correct starting pose for a given Cartesian motion?

* move_lin: It is using the current starting position of the EE, therefore it is always in its starting position for the linear motion.
* move_circ: Maybe we can use a linear motion from the current pose to the starting pose of move_circ motion. Another possibility is to use move it! to reach the starting position with a "PTP" motion.

ipa-fxm, 2016-01-12:
Most is done in #74, few has been separated into #69 and #75, rest is `WONTFIX` or `FIXONUSAGE`...:pointright: closing!

closed 2016-01-12 by ipa-fxm