4am-robotics/cob_control #62: Add configuration options for cob_omni_drive_controller/OdometryController

ipa-mdl, 2015-10-19:
* TF frames
* covariance

ipa-fxm, 2016-02-20:
TF frames - > https://github.com/ipa320/cob_control/blob/indigo_dev/cob_omni_drive_controller/src/odom_plugin.cpp#L47-L48
Where would you want the covariance to be parametrizable?

ipa-mdl, 2016-02-21:
The parameters of `OdometryTracker` can be passed in the constructor.
I have prepared covariance settings for pose and twist, this should be sufficient.

As far as I can see, there are 4 params needed:
* `parent_link` (="odom_combined")
* `child_link` (="base_footprint")
* `covariance_pose` (=0.1)
* `covariance_twist` (=0.1)

We don't need dynamic reconfigure for these.
(Dynamic reconfigure is interesting for the controller settings..)

ipa-fxm, 2016-02-21:
solved in #90 

Although, I think it would make sense to have the same configurability available for the `cob_undercarriage_ctrl_node` package, I think it is not worth to put any more effort into that package as it is not used anywhere and was just merged "as it was already implemented", right? @ipa-mdl @ipa-mig

ipa-fxm, 2016-02-21:
configurability for the twist_controller plugin is tracked in a separate issue #91

ipa-mig, 2016-02-22:
+1 **for not** porting the covariance configurability to `cob_undercarriage_ctrl_node` (as it hopefully will be deprecated soon)
+1 **for** porting the frame configurability to `cob_undercarriage_ctrl_node`, as this is something more general and important

About the covariances:
we need more parameters there.
Fully correct for 3D Pose would be 36, even though we ever only populate the diagonal entries (i.e. 6). And as we are 2D, 3 would actually be enough (i.e. `cov_xx`, `cov_yy` and `cov_yawyaw`).
Maybe, for our omnidirectional robots, `cov_xx==cov_yy`, but `cov_yawyaw` definitely needs to be something else.

maybe @ipa-srd could provide some input here about how we have them configured in the nav stack.

ipa-mdl, 2016-02-22:
Again. this just resembles https://github.com/ipa320/cob_driver/blob/indigo_dev/cob_undercarriage_ctrl/ros/src/cob_undercarriage_ctrl.cpp#L734
The covariance is static, this issue aims to make it a configurable option.
I guess `covariance_twist` is  `cov_yawyaw`?

ipa-mig, 2016-02-22:
True, but that doesn't make the old implementation correct ;-)

All I'm saying is that if we go through the hassle of making it configureable, we should make this right.

`cov_xx`, `cov_yy` and `cov_yawyaw` are just a more meaningful name for the diagonal entries (1, 2 and 6) of  `covariance_twist` and `covariance_pose` respectively.

Some more insights:
- The three components `xx`, `yy` and `yawyaw` (in theory together with `zz`, `rollroll` and `pitchpitch`, but I'm talking 2D from now on) can differ, so we should be able to set them independently. This holds for `covariance_twist` and `covariance_pose`.
- We assume here that we have no correlation between driving straight ahead or rotating, thus all non-diagonal (i.e. `xy`, `yx`, `rolly`, ...) entries are zero. This is probably not fully correct, but an acceptable error.
- `covariance_twist` is the covariance of the velocities (in `v_x`, `v_y` and `v_yaw`), i.e. the *certainty of the velocity*. This is something that could definitely be used.
- `covariance_pose` is the covariance of the pose of the robot (i.e. `x`, `y` and `yaw` with respect to the `odom_frame`), i.e. the *certainty of the base position in the `odom_frame`* . In theory, this grows without bounds (even though I have not seen it implemented like this somewhere)! Thus, this is actually something wich does not really make sense to use.
- `covariance_pose` will most probably depend on `covariance_twist` in a sense that the higher the values of the respective `covariance_twist` component is, the faster this component in `covariance_pose` grows.
- Thus, from my perspective, we could have `covariance_pose` all zeros (after we checked we actually don't use it somewhere...)

ipa-mdl, 2016-02-22:
Feel free to adapt it to your needs ;)

Related: https://github.com/ipa320/cob_driver/commit/4c7b86eb9a681b7f1a2a0e02df1aeb5b3e12ee0e

PR #90 (2016-03-04): configurable odometry_controller
fixes #62 

I chose to set the `frame_id` to "odom_wheel" by default because it's the odometry controller with input from the wheels only. "odom_combined" would be coming from a node combining odometry data from wheels as well as e.g. GPS or alike....
However, as "odom_combined" is the frame we use most of the time, we would need to set (at least) the `frame_id` parameter in all the `base_controller.yaml`...

(BTW, I vote for "odom_wheel" vs. "wheelodom" as it is more consistent with "odom_combined")

@ipa-mdl @ipa-mig please provide input!
closed 2016-03-04 by ipa-fxm