4am-robotics/cob_control #83: [cob_twist_controller] JointTrajectoryControllerInterface on Torso

ipa-fxm, 2016-02-09:
further investigate behavior of twist_controller when using ht JointTrajectoryControllerInterface

in #66 it seems that both the behavior of the utils MovingAverage as well as SimpsonIntegration are fine and perform acceptable
still during tests with hardware - torso in particular - resulting motion were quite jerky.

this can be seen in quite jerky `joint_trajectory_controller/state/desired` messages (__velocities__ as well as positions).
assumption is that the joint_trajectory_controller is not meant to be commanded at high update_rates (50Hz)
as the jerky motion has mainly be observed on Torso (Elmo), there also might be potential in firmware configuration and/or motor/sensor hardware optimization @ipa-mdl @ipa-tif @ipa-bdw

ipa-mdl, 2016-02-09:
I guess a spline interpolator is not the right controller for "high" rate control.
Have you tried the group forward controller with mode 7?
Did you turn off all velocity/acceleration limits in URDF/yaml? (https://github.com/ros-industrial/ros_canopen/issues/141)

ipa-fxm, 2016-02-10:
Note to myself:
Make plots comparing 
 - `q_dot_ik`
 - `q_dot_avg`
 - `derived_integrated_q_dot`
 - `jtc/state/desired/velocitites`
 - `joint_states/velocities`

Also, verify that `period` used in integrator is equal to `period` between two arriving twist commands #76

ipa-fxm, 2016-02-12:
Testing the period with the following LoC in `SimpsonIntegrator::updateIntegration()` and in `CobTwistController::solveTwist()`
```
ros::Time now = ros::Time::now();
ROS_INFO_STREAM("\t\tSIMPSON_INTEGRATOR: Period: " << (now - last_update_time_).toSec());
last_update_time_ = now;
```
shows that the integration/execution period are identical +/- 10^-4

ipa-fxm, 2016-02-12:
Here are some plots from using the joint_trajectory interface with the torso (commit 717608e54cd487ae7b555f06f6668db2c48606e5);
Publishing Twists at 10Hz
torso_2_joint
[image]
torso_3_joint
[image]

Publishing Twists at 50Hz:
torso_2_joint
[image]
torso_3_joint
[image]

bottom line: trajectory_interface is way too slow for...

ipa-fxm, 2016-02-12:
Plots using VelocityInterface (joint_group_velocity_controller) Mode3 at 50hz
torso_2_joint
[image]
torso_3_joint
[image]

limits as given in `torso_driver.yaml`:
```
  - name: torso_2_joint
    id: 32
    dcf_overlay: # "ObjectID": "ParameterValue" (both as strings)
      "6083": "6000" # profile acceleration, mgrad/sec^2
      "6084": "6000" # profile deceleration, mgrad/sec^2
      "60C5": "6000" # max acceleration, mgrad/sec^2
      "60C6": "6000" # max deceleration, mgrad/sec^2
  - name: torso_3_joint
    id: 33
    dcf_overlay: # "ObjectID": "ParameterValue" (both as strings)
      "6083": "80000" # profile acceleration, mgrad/sec^2
      "6084": "80000" # profile deceleration, mgrad/sec^2
      "60C5": "80000" # max acceleration, mgrad/sec^2
      "60C6": "80000" # max deceleration, mgrad/sec^2
```

ipa-fxm, 2016-02-12:
Plots using JointGroupInterpolatedPositionController (joint_group_position_controller) Mode 7:
torso_3_joint:
[image]

ipa-fxm, 2016-02-12:
Summary:
 - joint_trajectory_controller (mode7): internal interpolation seems to reduce the absolute magnitude of the velocities (would need to look into it in ros_controllers/joint_trajectory_controller)
 - joint_group_velocity_controller (mode3): somehow has internal acceleration limits and is not able to follow given (non-acceleration limited) velocity commands
 - joint_group_position_controller (mode7): position command seems to be "too jerky" and torso starts to vibrate...

ipa-fxm, 2016-02-12:
We found acceleration limits set wrong in the Elmo firmware (10000 for torso_3_joint) that explains the ramps as seen in the respective plot.
After setting the the firmware acceleration limits to 100000 the limits from the torso_driver.yaml (80000) are applied...and motion looks much more reactive
[image]

However, when switching back to trajectoy_mode, we always fall into diagnostics_error:
```
Full Name: /Actuators/Torso/torso torso_driver: torso_3_joint
Component: torso torso_driver: torso_3_joint
Hardware ID: none
Level: ERROR
Message: Node has emergency error; Motor has fault

error_register: 33
errors: 8611#0
```
and (several!) `recover` are needed...

ipa-fxm, 2016-02-12:
The above mentioned problem occurs when switching from mode 7 to mode 3, then moving the joint for more than 15° and then switch back from mode 7 to mode3...
A fix is to add the following to the defaults of torso_driver.yaml
```
  motor_layer:
    switching_state: 4 # switched on - stop controller before switching modes
```

:warning: In mode 3 end position limits are __not__ considered!

:warning: Also the movements have a significant overshoot 
  :point_right: needs more investigation - probably bad frame_tracking PID configuration...?

ipa-fxm, 2016-02-12:
ToDo's:
 1. investigate overshoot when using velocity_interface (mode3)
 2. investigate smoothing the commanded positions when using position_(interpol)_interface (mode7)
 3. investigate speedup/interpolation of joint_trajectory_interface (mode7)

ipa-fxm, 2016-02-19:
We have decided to go with the velocity_interface (mode3) for now as it yields the most promising performance...with have various temporary "hacks" for the still-missing joint_limit_interface....e.g. directly in elmo firmware as well as in the limiters of the twist_controller...

closed 2016-02-19 by ipa-fxm

ipa-mdl, 2016-02-19:
the new limit interface is tracked in #72