4am-robotics/cob_control #91: [cob_omni_drive_controller] Add configuration options for DiffDriveController

ipa-fxm, 2016-02-21:
separating this from #62 

 > (Dynamic reconfigure is interesting for the controller settings..)

ipa-fxm, 2016-02-21:
Looking into this...am I missing something?

For the bases already using the diff_drive_controller (as well as for simulation), in the base_controller.yaml files, the `max` are not set (e.g. [here](https://github.com/ipa320/cob_robots/blob/indigo_dev/cob_hardware_config/cob4-3/config/base_controller.yaml#L13-L14)) and thus default to [0.0](https://github.com/ipa320/cob_control/blob/indigo_dev/cob_omni_drive_controller/src/control_plugin.cpp#L28-L33) - __how can we actually drive then?__

ipa-fxm, 2016-02-21:
Maybe we could also start gathering a list of the parameters that should be made dynamically reconfigurable...@ipa-mdl @ipa-mig (and Nav-Team) please provide input...

ipa-mdl, 2016-02-21:
> Why are the limits commented in the yamls?

To move as fast as possible? ;)
Do we have to limit it?

ipa-mdl, 2016-02-21:
> gathering a list of the parameters that should be made dynamically reconfigurable...

Limits:
* `max_rot_velocity` (input limit)
* `max_trans_velocity` (input limit)
* `max_steer_rate` (output limit)
* `max_drive_rate` (output limit)

Params for the steer controller (could be even set per steer axis):
* `spring`
* `damp`
* `virt_mass`
* `d_phi_max`
* `dd_phi_max`

I am not sure if we really need to set these dynamically.

ipa-mig, 2016-02-22:
> > Why are the limits commented in the yamls?
> 
> To move as fast as possible? ;)
> Do we have to limit it?
> Update: it is just a limit for the input.

Well, from my point of view, it is **absolutely crucial** to limit the **output** to reasonable values.

Having no limit on the input is okay for me. But it might make sense to throw a warning, if it is something enormous (e.g. 5.0 m/s is something which we will not command, ever, to a service robot base), pointing to a bug in the node generating the commands. Thus having a high default for the input limits would be beneficial.

Also, from my perspective and in general:
-1 for making limits dynamic reconfigurable
+1 for  making controller variables dynamic reconfigurable

ipa-mdl, 2016-02-22:
> Well, from my point of view, it is absolutely crucial to limit the output to reasonable values.

This is just the old controller implementation. Feel free to extend limiting wherever you need it.

ipa-mig, 2016-02-22:
> * max_steer_rate (output limit)
> * max_drive_rate (output limit)

From this, I expected the output is already limited? So I'm happy there ;-)
This was just to give the full picture...

ipa-fxm, 2016-02-23:
As I see it, both input (`max_rot_velocity`, `max_trans_velocity`) __and__ output (`max_steer_rate`, `max_drive_rate`) limits are currently nowhere set within the yaml...thus defaulting to `0.0`...thus __not__ limiting neither input __nor__ output!

ipa-fxm, 2016-02-23:
Also, the structure of the yaml parser (with parsing the `defaults` tag) is not that ease to combine with the dynamic_reconfigure concept...(or I simply didn't fully get the templated-concept...)

ipa-fxm, 2016-02-23:
Not directly related to configurability, but likely related to configuration:
I don't know whether the behavior was similar before, but since switching to using the ros_canopen with the "old" cob4-2 platform, I observed some oscillation on the robot (to be seen in the odometry topic)...that even "sways" the whole upper body (torso) a bit...

ipa-fxm, 2016-02-24:
Here is the promised test run publishing a sine signal to the `angular/z` component of the `/base/twist_controller/command` topic (no smoother involved!):

50Hz
[image]

ipa-fxm, 2016-02-24:
After quick discussion with @ipa-mdl, also plot (on cob4-1):
 - `odometry/pose`
 - `base/joint_states

Compare behavior on raw (old Elmos + old undercarriage_ctrl)
Unfortunately, we do not have a cob4-3/4/6 (new Elmo + control_plugin) anymore and cob4-1 with old Elmos and old undercarriage_ctrl requires hardware modification.

ipa-fxm, 2016-02-24:
Step response of base:
[image]

ipa-fxm, 2016-03-05:
Discussion about base behavior after switching to new `ros_canopen` (starting in [this comment](https://github.com/ipa320/cob_control/issues/91#issuecomment-187657540)) has been answered and solved in #98

ipa-fxm, 2017-02-14:
Coming back - after the "UM 2 excursus" - to the [main discussion](https://github.com/ipa320/cob_control/issues/91#issuecomment-187653338) of this issue, i.e. (re-)configurability and limiting within the base_twist_controller:
Here is a detailed summary of current limiting behavior: https://github.com/ipa320/cob_control/issues/92#issuecomment-279648573

ipa-fxm, 2017-02-14:
Dynamic reconfigure for the default values of the SteerCtrl parameters, i.e. applied to **all** wheels, is implemented in #135

ipa-fxm, 2017-02-14:
**Plot Alarm**

Here are some plots using current (cob_control/cob_robots) `ipa320/indigo_dev` implementation/configuration using the test scripts provided in #134 performed in **gazebo simulation**:
(Commands are published directly on `/base/twist_controller/command`, i.e. no twist_mux, no velocity_smoother involved!)

### Step Response
1. Step Linear (wheels oriented in linear direction)
[image]
2. Step Linear (wheels oriented in rotational direction)
[image]
3. Step Rotation (wheels oriented in rotational direction)
[image]
4. Step Rotation (wheels oriented in linear direction)
[image]

### Sine Response (input publish frequency 10Hz)
1. Sine Linear (wheels oriented in linear direction)
[image]
2. Sine Linear (wheels oriented in rotational direction)
[image]
3. Sine Rotation (wheels oriented in rotational direction)
[image]
4. Sine Rotation (wheels oriented in linear direction)
[image]

### Sine Response (input publish frequency 50Hz)
2. Sine Linear (wheels oriented in rotational direction)
[image]
4. Sine Rotation (wheels oriented in linear direction)
[image]

ipa-fxm, 2017-02-14:
Some observations and more detailed analysis:

### Positioning of Steer Drives
To position the steer drives from a rotational direction to a linear direction or from a linear direction to a rotational direction, Currently takes about 0.5 seconds. And leads to a slight jolt in the odometry.
1.Rotational to Linear
[image]
[image]
2. Linear to Rotational
[image]
[image]

I think the steer drives could/should both **react** a bit faster and **level out** a bit faster, by tuning the SteerCtrl parameters

ipa-fxm, 2017-02-22:
configurability is added in #141

ipa-fxm, 2017-02-22:
closing this issue as #141 is merged

further activities and consequences, e.g. tuned default parameter, can be discussed in #143

closed 2017-02-22 by ipa-fxm