4am-robotics/cob_control #92: [cob_base_velocity_smoother] sawtooth behavior

ipa-fxm, 2016-02-24:
I (still) think this is worth creating an issue:

When investigating the base command pipeline (twist_mux -> velocity_smoother -> control_plugin -> odometry_plugin), I found the following behavior for the "new" velocity smoother.

Tested with a cpp-node that publishes a sine signal onto a `geometry_msgs::Twist` topic
`roslaunch cob_bringup velocity_smoother.launch robot:=cob4-1`

100Hz
[image]

50 Hz
[image]

30Hz
[image]

ipa-fxm, 2016-02-24:
What it does well is smoothing back to stop in case the publisher stops publishing
30Hz
[image]

ipa-fxm, 2016-02-24:
For comparison this is the behavor of the "old" cob_base_velocity_smoother (in it's default configuration):
`rosrun cob_base_velocity_smoother cob_base_velocity_smoother`

100Hz
[image]

50Hz
[image]

30Hz
[image]

ipa-fxm, 2016-02-24:
What the "old" smoother doesn't do well is smoothing back to 0.0:
[image]

ipa-fxm, 2016-02-24:
The "new" smoother's publish rate is set to 100.0 Hz
The "old" smoother's publish rate is set to 30.0 Hz

ipa-fxm, 2017-02-13:
After some discussions with @ipa-bnm and @ipa-mdl, I see the point in the current implementation of `cob_base_velocity_smoother` as it is not really supposed to "smooth-out" steps/jumps in the raw input signal but limit the input only in case it's surpassing either the velocity limit or the acceleration limit.
Otherwise it tries to follow the input signal as fast as possible - and as soon as the desired (input) velocity is reached keep publishing them.
Quoting @ipa-bnm:
 > Repeated publishing of the target velocity should not make a difference for the underlying base_twist_controller, as the input for the base_twist_controller would be the same if only published once...only that **with** the `cob_base_velocity_smoother` velocity and acceleration limits can be guaranteed on the Cartesian velocity!

If a real smoothing is desired/needed, the only possible way would be to introduce some lag, e.g. PT1 controller, as the "direction" of the input cannot be predicted.

ipa-fxm, 2017-02-13:
Basically the  question is:
Are we ok with the described non-continuous "sawtooth" behaviour in the input signal of the base_twist_controller?
The cob_base_velocity_smoother then would rather act as a limiter interface operating on the Cartesian velocity level.
Maybe we could add this input limiting (i.e. acceleration limiting) to the base_twist_controller directly - saving one subscribe-publish-cycle

Or do we need something continuous, i.e. "stetig", as input for the base_twist_controller?
I guess atm only input Cartesian and output joint velocities are limitable in base_twist_controller.

ipa-mdl, 2017-02-13:
> I guess atm only input Cartesian and output joint velocities are limitable in base_twist_controller.

Only output velocities get limited.
Drive velolcity by velocity limit, steer velocity by velocity and acceleration limit.

> Are we ok with the described non-continuous "sawtooth" behaviour in the input signal of the base_twist_controller?

Can you prepare plots of the controllers response?

ipa-fxm, 2017-02-13:
I think we definitely need to solve this together with "tuning" the configuration of the base_twist_controller...
I will prepare configurability (see https://github.com/ipa320/cob_control/issues/91) and plots tomorrow...

ipa-fxm, 2017-02-14:
> Only output velocities get limited.
 > Drive velocity by velocity limit, steer velocity by velocity and acceleration limit.

@ipa-mdl @ipa-mig @ipa-bnm FYI

Here is some information about limiting I found in the base_twist_controller:
 - [`limitValue`](https://github.com/ipa320/cob_control/blob/indigo_dev/cob_omni_drive_controller/src/UndercarriageCtrlGeom.cpp#L185-L194) function is used throughout all the following limiting...
     It's "croping" the `value` to the interval [`-limit`, `limit`] in case `limit!=0`.
- Input Cartesian Velocity
   - translational velocity, i.e. `twist.linear.x`, `twist.linear.y` is limited [here](https://github.com/ipa320/cob_control/blob/indigo_dev/cob_omni_drive_controller/src/control_plugin.cpp#L143-L144) using [`max_trans_velocity`](https://github.com/ipa320/cob_control/blob/indigo_dev/cob_omni_drive_controller/src/control_plugin.cpp#L36-L40) which defaults to `0.0`, i.e. unlimited (currently this parameter is commented in the [base_controller.yaml](https://github.com/ipa320/cob_robots/blob/indigo_dev/cob_hardware_config/cob4-2/config/base_controller.yaml#L14)
   - rotational velocity, i.e. `twist.angular.z` is limited [here](https://github.com/ipa320/cob_control/blob/indigo_dev/cob_omni_drive_controller/src/control_plugin.cpp#L145) using [`max_rot_velocity`](https://github.com/ipa320/cob_control/blob/indigo_dev/cob_omni_drive_controller/src/control_plugin.cpp#L31-L35) which defaults to `0.0`, i.e. unlimited (currently this parameter is commented in the [base_controller.yaml](https://github.com/ipa320/cob_robots/blob/indigo_dev/cob_hardware_config/cob4-2/config/base_controller.yaml#L13)
   - Note that the defaults for `max_trans_velocity` and `max_rot_velocity` are different in [old undercarriage_ctrl](https://github.com/ipa320/cob_driver/blob/indigo_dev/cob_undercarriage_ctrl/ros/src/cob_undercarriage_ctrl.cpp#L175-L194) and [new undercarriage_ctrl_node](https://github.com/ipa320/cob_control/blob/indigo_dev/cob_undercarriage_ctrl_node/src/cob_undercarriage_ctrl_new.cpp#L165-L184)
 - Output Joint Velocities/Acceleration
   - steer/drive velocity is limited [here](https://github.com/ipa320/cob_control/blob/indigo_dev/cob_omni_drive_controller/src/UndercarriageCtrlGeom.cpp#L226-L227) using [`max_steer_rate`/`max_drive_rate`](https://github.com/ipa320/cob_control/blob/indigo_dev/cob_omni_drive_controller/src/param_parser.cpp#L66-L67) which defaults to `0.0`, i.e. unlimited (currently this parameter is not set in the [base_controller.yaml](https://github.com/ipa320/cob_robots/blob/indigo_dev/cob_hardware_config/cob4-2/config/base_controller.yaml)
   - steer velocity is also limited limited [here](https://github.com/ipa320/cob_control/blob/indigo_dev/cob_omni_drive_controller/src/UndercarriageCtrlGeom.cpp#L219-L220) using [`d_phi_max`](https://github.com/ipa320/cob_control/blob/indigo_dev/cob_omni_drive_controller/src/param_parser.cpp#L78) which is required to be set and currently set to `10.0` in the [base_controller.yaml]https://github.com/ipa320/cob_robots/blob/indigo_dev/cob_hardware_config/cob4-2/config/base_controller.yaml#L22)
   - steer acceleration is limited [here](https://github.com/ipa320/cob_control/blob/indigo_dev/cob_omni_drive_controller/src/UndercarriageCtrlGeom.cpp#L216-L217) using [`dd_phi_max`](https://github.com/ipa320/cob_control/blob/indigo_dev/cob_omni_drive_controller/src/param_parser.cpp#L79) which is required to be set and currently set to `40.0` in the [base_controller.yaml]https://github.com/ipa320/cob_robots/blob/indigo_dev/cob_hardware_config/cob4-2/config/base_controller.yaml#L23)

ipa-mdl, 2017-02-14:
Not sure why we have `d_phi_max` AND `max_steer_rate`..

ipa-fxm, 2017-02-14:
Here is a comparison of using the `cob_base_velocity_smoother` vs. **not** using the `cob_base_velocity_smoother` (ipa320/indigo_dev implementation/configuration):

1. Step Linear 10Hz (wheels oriented in rotational direction) published to base_twist_controller directly
[image]
2. Step Linear 10Hz (wheels oriented in rotational direction) published to velocity_smoother 50Hz
[image]
3. Step Rotation 10Hz (wheels oriented in linear direction) published to base_twist_controller directly
[image]
4. Step Rotation 10Hz (wheels oriented in linear direction) published to velocity_smoother 50Hz
[image]