4am-robotics/cob_control #116: cob_cartesian_controller does not build on indigo armhf

tfoote, 2016-11-25:
Example: http://build.ros.org/view/Ibin_arm_uThf/job/Ibin_arm_uThf__cob_cartesian_controller__ubuntu_trusty_armhf__binary/222/console

```
04:05:19 [100%] Building CXX object CMakeFiles/cartesian_controller_node.dir/src/cartesian_controller_node.cpp.o
04:05:19 /usr/bin/arm-linux-gnueabihf-g++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"cob_cartesian_controller\" -g -O2 -fstack-protector --param=ssp-buffer-size=4 -Wformat -Werror=format-security -DNDEBUG -D_FORTIFY_SOURCE=2  -I/tmp/binarydeb/ros-indigo-cob-cartesian-controller-0.6.14/obj-arm-linux-gnueabihf/devel/include -I/tmp/binarydeb/ros-indigo-cob-cartesian-controller-0.6.14/include -I/opt/ros/indigo/include    -o CMakeFiles/cartesian_controller_node.dir/src/cartesian_controller_node.cpp.o -c /tmp/binarydeb/ros-indigo-cob-cartesian-controller-0.6.14/src/cartesian_controller_node.cpp
04:06:25 Linking CXX executable devel/lib/cob_cartesian_controller/cartesian_controller_node
04:06:25 /usr/bin/cmake -E cmake_link_script CMakeFiles/cartesian_controller_node.dir/link.txt --verbose=1
04:06:25 /usr/bin/arm-linux-gnueabihf-g++   -g -O2 -fstack-protector --param=ssp-buffer-size=4 -Wformat -Werror=format-security -DNDEBUG -D_FORTIFY_SOURCE=2     CMakeFiles/cartesian_controller_node.dir/src/cartesian_controller_node.cpp.o  -o devel/lib/cob_cartesian_controller/cartesian_controller_node -rdynamic devel/lib/libcartesian_controller.so /opt/ros/indigo/lib/libtf.so /opt/ros/indigo/lib/libtf2_ros.so /opt/ros/indigo/lib/libactionlib.so /opt/ros/indigo/lib/libmessage_filters.so /opt/ros/indigo/lib/libroscpp.so -lboost_signals -lboost_filesystem /opt/ros/indigo/lib/libxmlrpcpp.so /opt/ros/indigo/lib/libtf2.so /opt/ros/indigo/lib/librosconsole.so /opt/ros/indigo/lib/librosconsole_log4cxx.so /opt/ros/indigo/lib/librosconsole_backend_interface.so -llog4cxx -lboost_regex /opt/ros/indigo/lib/libroscpp_serialization.so /opt/ros/indigo/lib/librostime.so -lboost_date_time /opt/ros/indigo/lib/libcpp_common.so -lboost_system -lboost_thread -lpthread -lconsole_bridge devel/lib/libtrajectory_interpolator.so devel/lib/libprofile_generator.so devel/lib/libcartesian_controller_utils.so /opt/ros/indigo/lib/libtf.so /opt/ros/indigo/lib/libtf2_ros.so /opt/ros/indigo/lib/libactionlib.so /opt/ros/indigo/lib/libmessage_filters.so /opt/ros/indigo/lib/libroscpp.so -lboost_signals -lboost_filesystem /opt/ros/indigo/lib/libxmlrpcpp.so /opt/ros/indigo/lib/libtf2.so /opt/ros/indigo/lib/librosconsole.so /opt/ros/indigo/lib/librosconsole_log4cxx.so /opt/ros/indigo/lib/librosconsole_backend_interface.so -llog4cxx -lboost_regex /opt/ros/indigo/lib/libroscpp_serialization.so /opt/ros/indigo/lib/librostime.so -lboost_date_time /opt/ros/indigo/lib/libcpp_common.so -lboost_system -lboost_thread -lpthread -lconsole_bridge -Wl,-rpath,/tmp/binarydeb/ros-indigo-cob-cartesian-controller-0.6.14/obj-arm-linux-gnueabihf/devel/lib:/opt/ros/indigo/lib: 
04:06:26 devel/lib/libprofile_generator.so: undefined reference to `vtable for TrajectoryProfileRamp'
04:06:26 devel/lib/libprofile_generator.so: undefined reference to `vtable for TrajectoryProfileSinoid'
04:06:26 collect2: error: ld returned 1 exit status
04:06:26 make[4]: *** [devel/lib/cob_cartesian_controller/cartesian_controller_node] Error 1
04:06:26 make[4]: Leaving directory `/tmp/binarydeb/ros-indigo-cob-cartesian-controller-0.6.14/obj-arm-linux-gnueabihf'
04:06:26 make[3]: *** [CMakeFiles/cartesian_controller_node.dir/all] Error 2
04:06:26 make[3]: Leaving directory `/tmp/binarydeb/ros-indigo-cob-cartesian-controller-0.6.14/obj-arm-linux-gnueabihf'
04:06:26 make[2]: *** [all] Error 2
04:06:26 make[2]: Leaving directory `/tmp/binarydeb/ros-indigo-cob-cartesian-controller-0.6.14/obj-arm-linux-gnueabihf'
04:06:26 dh_auto_build: make -j1 returned exit code 2
04:06:26 make[1]: *** [override_dh_auto_build] Error 2
04:06:26 make[1]: Leaving directory `/tmp/binarydeb/ros-indigo-cob-cartesian-controller-0.6.14'
04:06:26 make: *** [build] Error 2
04:06:26 dpkg-buildpackage: error: debian/rules build gave error exit status 2
04:06:26 E: Building failed
```

ipa-fmw, 2016-11-25:
@ipa-mdl @ipa-fxm: any hint?

ipa-mdl, 2016-11-26:
> undefined reference to `vtable for TrajectoryProfileRamp`

missing virtual destructor in base class?

ipa-mdl, 2016-11-26:
The inline declarations https://github.com/ipa320/cob_control/blob/indigo_dev/cob_cartesian_controller/src/trajectory_profile_generator/trajectory_profile_generator_ramp.cpp#L34 look suspicious.

ipa-fxm, 2016-12-15:
@tfoote @ipa-mdl 
If I work on this, how can I test this?

tfoote, 2016-12-16:
@ipa-fxm If you install docker and qemu you can use docker to get an armhf environment: 
```
$ docker run -ti --rm osrf/ubuntu_armhf:trusty bash
root@e7e775490ae5:/#
```

This is stock ubuntu you'll need to do the standard install and then setup your workspace and compile in there. I think it's likely just dropping that keyword in the cpp.

ipa-fxm, 2016-12-16:
Hmm, I'm no docker expert, but this doesn't seem right:
```
$ docker run -ti --rm osrf/ubuntu_armhf:trusty bashpanic: standard_init_linux.go:175: exec user process caused "exec format error" [recovered]
	panic: standard_init_linux.go:175: exec user process caused "exec format error"

goroutine 1 [running, locked to thread]:
panic(0x7ddea0, 0xc8201117a0)
	/usr/local/go/src/runtime/panic.go:481 +0x3e6
github.com/urfave/cli.HandleAction.func1(0xc8200e52e8)
	/go/src/github.com/opencontainers/runc/Godeps/_workspace/src/github.com/urfave/cli/app.go:478 +0x38e
panic(0x7ddea0, 0xc8201117a0)
	/usr/local/go/src/runtime/panic.go:443 +0x4e9
github.com/opencontainers/runc/libcontainer.(*LinuxFactory).StartInitialization.func1(0xc8200e4bf8, 0xc82001e0c8, 0xc8200e4d08)
	/go/src/github.com/opencontainers/runc/Godeps/_workspace/src/github.com/opencontainers/runc/libcontainer/factory_linux.go:259 +0x136
github.com/opencontainers/runc/libcontainer.(*LinuxFactory).StartInitialization(0xc82005d630, 0x7fe990809728, 0xc8201117a0)
	/go/src/github.com/opencontainers/runc/Godeps/_workspace/src/github.com/opencontainers/runc/libcontainer/factory_linux.go:277 +0x5b1
main.glob.func8(0xc82007aa00, 0x0, 0x0)
	/go/src/github.com/opencontainers/runc/main_unix.go:26 +0x68
reflect.Value.call(0x7448a0, 0x8f0bd0, 0x13, 0x839998, 0x4, 0xc8200e5268, 0x1, 0x1, 0x0, 0x0, ...)
	/usr/local/go/src/reflect/value.go:435 +0x120d
reflect.Value.Call(0x7448a0, 0x8f0bd0, 0x13, 0xc8200e5268, 0x1, 0x1, 0x0, 0x0, 0x0)
	/usr/local/go/src/reflect/value.go:303 +0xb1
github.com/urfave/cli.HandleAction(0x7448a0, 0x8f0bd0, 0xc82007aa00, 0x0, 0x0)
	/go/src/github.com/opencontainers/runc/Godeps/_workspace/src/github.com/urfave/cli/app.go:487 +0x2ee
github.com/urfave/cli.Command.Run(0x83c828, 0x4, 0x0, 0x0, 0x0, 0x0, 0x0, 0x8d0180, 0x51, 0x0, ...)
	/go/src/github.com/opencontainers/runc/Godeps/_workspace/src/github.com/urfave/cli/command.go:191 +0xfec
github.com/urfave/cli.(*App).Run(0xc820001b00, 0xc82000a100, 0x2, 0x2, 0x0, 0x0)
	/go/src/github.com/opencontainers/runc/Godeps/_workspace/src/github.com/urfave/cli/app.go:240 +0xaa4
main.main()
	/go/src/github.com/opencontainers/runc/main.go:137 +0xe24
```
I [installed]() docker for Ubuntu and ```sudo apt-get install qemu```
The hello-world example reports success...

ipa-fxm, 2016-12-16:
I committed the suggested changes to a [branch](https://github.com/ipa-fxm/cob_control/tree/fix_cartesian_controller_armhf) in case someone can test this...PR is #123

ipa-fxm, 2017-01-19:
I merged #123 but was unable to test with docker whether the provided fix actually solves this issue...
Need to see on next release

ipa-fxm, 2017-07-20:
fixed according to <http://build.ros.org/job/Kbin_djv8_dJv8__cob_cartesian_controller__debian_jessie_arm64__binary/2/display/redirect>

closed 2017-07-20 by ipa-fxm