4am-robotics/cob_control #129: [cleanup] use ros logging

ipa-fxm, 2017-01-20:
There are quite some nodes using python2 `print`
```
./cob_twist_controller/scripts/test/test_raw31.py
./cob_twist_controller/scripts/test/test_careobot_base.py
./cob_twist_controller/scripts/test_publisher_twist_series.py
./cob_twist_controller/scripts/test_publisher_twist_sine.py
./cob_twist_controller/scripts/evaluate_dbg_jnt_velocity_tests.py
./cob_twist_controller/scripts/test_publisher_vel.py
./cob_twist_controller/src/data_collection/data_collection.py
./cob_obstacle_distance/scripts/test_interactive_obstacle_node.py
./cob_frame_tracker/scripts/interactive_frame_target.py
./cob_cartesian_controller/scripts/test_move_lin.py
./cob_cartesian_controller/scripts/test_move_circ.py
./cob_cartesian_controller/src/simple_cartesian_interface/simple_cartesian_interface.py
./cob_model_identifier/scripts/input_series_twist.py
```
use proper `rospy.logXXX` (or at least python3-compatible `print()`)

Also get rid of `std::cout`:
```
./cob_undercarriage_ctrl_node/src/cob_undercarriage_ctrl_new.cpp
./cob_trajectory_controller/common/include/cob_trajectory_controller/RefValJS_PTP.h
./cob_model_identifier/src/output_recorder.cpp
./cob_collision_velocity_filter/src/cob_collision_velocity_filter.cpp
```

ipa-fxm, 2017-01-20:
This might also be the case in other repos...

ipa-mig, 2017-01-20:
To enforce proper usage of `print` (if using it at all) in python, there exists:

```Python
form __future__ import print_function
```

See http://wiki.ros.org/python_2_and_3_compatible_code#Print_is_now_a_function