4am-robotics/cob_control #131: [cob_footprint_observer] provide a convex polygon as footprint, not a rectangle

ipa-mig, 2017-01-27:
Currently, the cob_footprint_observer only outputs a rectangular footprint. This should be adapted to provide a convex polygon.

ipa-fxm, 2017-02-09:
For generating a non-rectangular footprint, i.e. convexHull for a given set of (2D-projected) (tf-Frame) points, we could use [cv::convexHull](http://docs.opencv.org/trunk/d3/dc0/group__imgproc__shape.html#ga014b28e56cb8854c0de4a211cb2be656). It also provides [cv::minAreaRect](http://docs.opencv.org/trunk/d3/dc0/group__imgproc__shape.html#ga3d476a3417130ae5154aea421ca7ead9) and [cv::minEnclosingCircle](http://docs.opencv.org/trunk/d3/dc0/group__imgproc__shape.html#ga8ce13c24081bbc7151e9326f412190f1)

ipa-fxm, 2017-02-09:
Still investigating possibilities to how to get points, or footprint, properly considering collision meshes