4am-robotics/cob_control #139: [kinetic][cob_collision_velocity_filter] Could not get robot pose

ipa-fxm, 2017-02-17:
When testing cob_bringup on cob4-8 under kinetic, I get the following warning messages **once per second** (copied from rqt_console):

```
Node: /collision_velocity_filter
Time: 14:38:41.047085207 (2017-02-17)
Severity: Warn
Published Topics: /base/twist_mux/command_safe, /collision_velocity_filter/anti_collision_costmap/costmap, /collision_velocity_filter/anti_collision_costmap/costmap_updates, /collision_velocity_filter/anti_collision_costmap/footprint, /collision_velocity_filter/anti_collision_costmap/obstacle_layer/clearing_endpoints, /collision_velocity_filter/anti_collision_costmap/obstacle_layer/parameter_descriptions, /collision_velocity_filter/anti_collision_costmap/obstacle_layer/parameter_updates, /collision_velocity_filter/anti_collision_costmap/parameter_descriptions, /collision_velocity_filter/anti_collision_costmap/parameter_updates, /collision_velocity_filter/parameter_descriptions, /collision_velocity_filter/parameter_updates, /collision_velocity_filter/velocity_limited_marker, /relevant_obstacles_grid, /rosout

Could not get robot pose, cancelling reconfiguration

Location:
/tmp/binarydeb/ros-kinetic-costmap-2d-1.14.0/src/costmap_2d_ros.cpp:Costmap2DROS::movementCB:355
```

```
Node: /collision_velocity_filter
Time: 14:38:40.959705538 (2017-02-17)
Severity: Warn
Published Topics: /base/twist_mux/command_safe, /collision_velocity_filter/anti_collision_costmap/costmap, /collision_velocity_filter/anti_collision_costmap/costmap_updates, /collision_velocity_filter/anti_collision_costmap/footprint, /collision_velocity_filter/anti_collision_costmap/obstacle_layer/clearing_endpoints, /collision_velocity_filter/anti_collision_costmap/obstacle_layer/parameter_descriptions, /collision_velocity_filter/anti_collision_costmap/obstacle_layer/parameter_updates, /collision_velocity_filter/anti_collision_costmap/parameter_descriptions, /collision_velocity_filter/anti_collision_costmap/parameter_updates, /collision_velocity_filter/parameter_descriptions, /collision_velocity_filter/parameter_updates, /collision_velocity_filter/velocity_limited_marker, /relevant_obstacles_grid, /rosout

Costmap2DROS transform timeout. Current time: 1487338720.9596, global_pose stamp: 0.0000, tolerance: 0.3000

Location:
/tmp/binarydeb/ros-kinetic-costmap-2d-1.14.0/src/costmap_2d_ros.cpp:Costmap2DROS::getRobotPose:538
```

@ipa-mig @ipa-fmw @ipa-nhg 
I see the `global_pose.stamp` is `0.0`, but I don't know why...
Do I need to localize or do something to set the robot_pose?
Is this a wrong configuration of cob4-8?

Hints and help appreciated!

ipa-fxm, 2017-07-10:
@ipa320/navigation-push 
I'm testing kinetic and came across this Warning again...
Though, it's just a warning and so far I could not figure out what the effects might be...still, it's pretty "spammy"...thus, we should try to avoid it...but I do not have a clue...any ideas?

ipa-fxm, 2017-07-24:
related issue is https://github.com/ros-planning/navigation/issues/517

fmessmer, 2019-05-29:
the costmap parameters were just wrong, see: https://github.com/mojin-robotics/mojin_robots/pull/167

closed 2019-06-05 by fmessmer

Shentheman, 2019-06-10:
Hi @fmessmer:
Can this issue be resolved by changing the costmap parameters at https://github.com/ipa320/cob_robots/blob/indigo_dev/cob_hardware_config/robots/raw3-5/config/local_costmap_params.yaml? Could you share the new costmap parameters if possible?

Thank you!

fmessmer, 2019-07-29:
I guess the main trick was to set the following in `local_costmap_params.yaml`
```
- global_frame: /base_link
- robot_base_frame: /base_footprint
+ global_frame: base_footprint
+ robot_base_frame: base_footprint
```