4am-robotics/cob_control #152: [cob_twist_controller] refactor constraints code

ipa-fxm, 2017-04-06:
- [] remove priority template: `<typename PRIO = uint32_t>`
 - [] separate and clearly name constraint builder/factory from actual constraints
 - [] use *.cpp and create library instead of *_impl.h
 - [] extract abstract GPM solver (i.e. just GPM formular) from actual GPM implementation (i.e. sum over constraint list)

 - [] remove dynamic reconfigurability for solver types

ipa-fxm, 2017-04-07:
@ipa-fxm-cm 
I think we should keep the dynamic reconfigurability for the time being...so could you please focus on "remove priority template" first?

ipa-bfb, 2017-04-07:
I agree. At the moment for me at least dynamic reconfigure is really useful. But once we finish all this solvers and make them stable we can start removing it.

ipa-fxm, 2017-04-07:
In https://github.com/ipa320/cob_control/pull/126 we started to debug #125, but did not pursue this any further....thus it has been closed unmerged...

ipa-fxm, 2017-04-13:
getting rid of the priority template is not that easy....I started it in [my branch](https://github.com/ipa-fxm/cob_control/tree/remove_prio_template), but I'm not able to compile it yet...