4am-robotics/cob_control #154: Compability with ur_modern_driver

ipa-bfb, 2017-04-13:
I successfully tested the new ur-modern_driver which allow us to send speed commands to the UR arm.

However when I was trying to use it with twist controller I got the following:

[ERROR] [1492072763.263901255]: Client [/arm/ur_driver] wants topic /arm/joint_group_velocity_controller/command to have datatype/md5sum [trajectory_msgs/JointTrajectory/65b4f94a94d1ed67169da35a02f33d3f], but our version has [std_msgs/Float64MultiArray/4b7d974086d4060e7db4613a7e6c3ba4]. Dropping connection.

What do you recommend to do? Should we change the type in twist controller of the publish topic?

ipa-fxm, 2017-04-13:
Well, it seems that `ur_modern_driver` has a `joint_trajectory_controller` loaded to the `controller_manager` but the name has been set to `/arm/joint_group_velocity_controller`.
Please check the type of the topic the `ur_modern_controller` subscribes to (in the code).

If `ur_modern_driver` subscribes to a `trajectory_msgs/JointTrajectory` topic, you could

ipa-fxm, 2017-04-13:
What controller did you load for the `ur_modern_driver`?
A `joint_trajectory_controller` subscribing to `trajectory_msgs/JointTrajectory` or a `joint_group_velocity_controller` subscribing to `std_msgs/Float64MultiArray`.
Seems like you mapped the topic name of a `joint_trajectory_controller` to match the `joint_group_velocity_controller`. This won't work.

Either you load a proper `joint_group_velocity_controller` to the controller_manager of the `ur_modern_driver``- then you can use the `cob_twist_controller/ControllerInterfaceVelocity` controller interface. 
Or you stick to the `joint_trajectory_controller` being loaded to the controller_manager of the `ur_modern_driver` - but then you would need to set the controller interface of the twist controller to `cob_twist_controller/ControllerInterfaceTrajectory`...but note: this interface is not as well tested as the JointGroupVelocity interface...

ipa-bfb, 2017-04-13:
Solved ;) 

You are the best solving my issues xD

Have a nice Easter ;)

closed 2017-04-13 by ipa-fxm