4am-robotics/cob_control #175: cob_cartesian_controller: How to start ?

Ghost-CBC, 2018-01-20:
My robot is Schunk_lwa4d, I can move it use the dashboard interface.
Ｔｈｅｎ I roslaunch [cartesian_controller.launch] and rosrun [move_lin.py](in schunk_lwa4d package),
the line [sss.move("arm"...] can move the arm, but the line [sci.move_lin(pose, "arm_base_link", profile)] failed, the terminal showed as follow:
1. [WARN] [WallTime: 1516411841.868259] Waiting for ActionServer: /cartesian_trajectory_action
2. [ERROR] [WallTime: 1516411845.868698] ActionServer not available within timeout

I don't know how to solve this.
I hope you can help me.
Thanks in advance.

Ghost-CBC, 2018-01-20:
@ipa-fxm Sir, could you give me some advice?

ipa-fxm, 2018-02-14:
There is a [`cartesian_controller.launch`](https://github.com/ipa320/schunk_robots/blob/indigo_dev/schunk_lwa4d/launch/cartesian_controller.launch) in the `schunk_lwa4d` package that can be used...

Please note, `cob_cartesian_controller` is WIP and might not be fully functional.

I'll close this issue here....@Ghost-CBC please post further problems with `schunk_lwa4d` in the respective issue tracker https://github.com/ipa320/schunk_robots/issues

closed 2018-02-14 by ipa-fxm