4am-robotics/cob_control #187: [cob_omni_drive_controller] controller keeps sending commands when it shouldn't

fmessmer, 2018-06-21:
I happened to come across this issue today when test driving paul-unity around with the joystick:

I saw the back_rotation joint kept moving slowly although I was not commanding anymore (I think mainly because the driver also went into QuickStop state, but I don't have proof for this). Interesting thing is the following plot where you can see, **that there is still a non-zero wheel-command for the respective back_rotation joint!!!**

[image]

I could also see the spikes on the velocity joint_states for various base joints:
[image]

For the rest of the time velocity joint_states is following the wheel_commands/target_velocity quite well:
[image]

related to https://github.com/mojin-robotics/cob4/issues/684

fmessmer, 2018-06-21:
@ipa-mdl 
can you think of a case where this could happen?
I mean generating a wheel_command/target_velocity where just the back_rotation joint is moving seems **very unlikely** (see also the steep drop of all the other target_velocities - deadman released)

could this also be an issue with ros_canopen (we are using latest kinetic release)?

bvanelli, 2019-02-11:
Hello, please let me know if he are having the same problem. You stated that the the joints keep moving even though no commands are being issued. I noticed that when the simulation starts the robot begins to spin in place even though nothing is being sent into `/base/velocity_smoother/command` or `/base/twist_controller/command` topics. I tried to instead send zeroes into the topics and the velocity goes back to zero! (see picture)

[image]