4am-robotics/cob_control #241: Noetic release

Timple, 2020-06-16:
Would you mind releasing this package also for the newest distribution ROS Noetic? :)

fmessmer, 2020-06-16:
what package are you most interested in?
we haven't worked on a proper Noetic release for any of the `cob_X` repos - except for `cob_extern`
so there might be a couple of upstream dependencies missing

Timple, 2020-06-17:
In one of our bots we use **cob_base_velocity_smoother**. Also we're evaluating the **cob_collision_velocity_filter** at the moment. So those two would be nice :)

fmessmer, 2020-06-17:
I don't know when I will find the time for a proper Noetic release - and I don't want to do a partial release - when I start with it I'll do it for the whole cob-world
but the packages should already be Python3 compatible - thus you should at least be able to use them from source....

Timple, 2020-06-18:
That's fine! For now we can work from source. Consider this a placeholder issue then.

fmessmer, 2020-10-21:
done: https://github.com/ros/rosdistro/pull/27065

closed 2020-10-21 by fmessmer